Kawasaki-Robotics/khi_robot #1: Robot xacros not macros

gavanderhoorn, 2019-03-29:
Thank you very much for releasing ROS support for Kawasaki robots :bowing_man: :+1: 

While looking at the [rs_description](https://github.com/Kawasaki-Robotics/khi_robot/blob/cbfa6eac6a313ac281135328f00d304fabfc3fe4/rs_description) package, I noticed that the `.urdf.xacro` files do not include any [`xacro:macro`s](http://wiki.ros.org/xacro#Macros) of the robots ([this one](https://github.com/Kawasaki-Robotics/khi_robot/blob/cbfa6eac6a313ac281135328f00d304fabfc3fe4/rs_description/urdf/rs007l.urdf.xacro) for the RS007 fi).

This makes it difficult to use the robot model in a larger scene, as it cannot be easily composed with other xacros.

Would you consider merging a Pull Request that transforms the `.urdf.xacro`s into `xacro:macro`s? If the macros are then stored in a separate file, the `.urdf.xacro` could `xacro:include` those and then call the macro (and thereby instantiate the model).

The public API of `rs_description` would not change, but by including the file that contains the `xacro:macro`, using the Kawasaki robot models in a larger work cell would become possible.

gavanderhoorn, 2019-03-29:
As an example for discussion: https://github.com/Kawasaki-Robotics/khi_robot/compare/master...gavanderhoorn:add_xacro_macros.

matsui-hiro, 2019-04-02:
Thank you for reporting the issue.
As you say, it is difficult to model multiple robots with the same cell if they are not xacro: macro.
I will adopt your plan.

gavanderhoorn, 2019-04-03:
#2 was merged, so this has been addressed.

A follow-up PR to make the xacros really composable will be submitted.

closed 2019-04-03 by gavanderhoorn