Kawasaki-Robotics/khi_robot #3: Targeted ROS versions

gavanderhoorn, 2019-04-01:
The readme states:

> ROS distribution `Kinetic` is supported.

but the MoveIt configuration and the `_description` packages give the impression that Indigo was/is also supported.

Should Indigo support be maintained?

matsui-hiro, 2019-04-02:
Indigo is not supported because we evaluated  for actual robot only with Kinetic.
I think that the simulator can be used with also Indigo but didn't try yet.
I'm sorry that there is no support plan for now.

gavanderhoorn, 2019-04-02:
> Indigo is not supported because we evaluated for actual robot only with Kinetic.
> I think that the simulator can be used with also Indigo but didn't try yet.

Clear. Thanks.

> I'm sorry that there is no support plan for now.

No need to apologise. I don't use Indigo, but just wanted to make sure before changing things to be compatible with Kinetic (and newer) releases only.

closed 2019-04-02 by gavanderhoorn