Kawasaki-Robotics/khi_robot #4: Suggestion: prefix all pkgs with 'khi_' (REP-144 compliance)

gavanderhoorn, 2019-04-01:
Some packages already have this prefix (such as `khi_robot_control`), but the suggestion is to prefix *all* packages in this repository with `khi_` (so `rs_description` becomes: `khi_rs_description`, `duaro_description` becomes `khi_duaro_description`, etc).

There are two main motivations for doing this:

 1. avoid name clashes: prefixing with `khi_` reduces the chance of a(n unrelated) package with the exact same name to near zero (ROS package names exist in a single, flat shared namespace and name clashes do happen)
 1. package provenance is immediately clear: the `khi_` prefix uniquely identifies these packages are coming from Kawasaki Heavy Industries

gavanderhoorn, 2019-04-01:
It would also make these packages more compliant with [REP-144: ROS Package Naming](http://www.ros.org/reps/rep-0144.html) (specifically: *Naming Rules*).

d-nakamichi, 2019-04-03:
OK, we'll take the suggestion: `khi_` prefix.

gavanderhoorn, 2019-04-10:
This would seem to be addressed with the merge of #10.

closed 2019-04-10 by gavanderhoorn