Kawasaki-Robotics/khi_robot #17: Avoid suggesting users to run nodes as 'root' user

gavanderhoorn, 2019-04-19:
In `docs/ConnectingRealRobot.md`, users are currently advised to run the bringup `.launch` files as the super user (ie: `root`) ([here](https://github.com/Kawasaki-Robotics/khi_robot/blame/09ff11b343491ed98109d03a14b722ef8660775a/docs/ConnectingRealRobot.md#L27
)).

In general starting ROS nodes (and `.launch` files) as `root` is not recommended, and it is also typically not needed.

Suggestion: use the approach documented [here](https://frankaemika.github.io/docs/installation.html#allow-a-user-to-set-real-time-permissions-for-its-processes), where a new group is created that gets sufficient privileges to set real-time scheduling priorities (among other things) without being `root` (using `/etc/security/limits.conf`).

d-nakamichi, 2019-04-23:
Thanks for your suggestion.
OK, I'll add description of real-time configuration using `limits.conf`.

k-okada, 2019-04-26:
fyi another approach -> https://github.com/tork-a/minas/blob/master/minas_control/CMakeLists.txt#L42-L53

gavanderhoorn, 2019-05-08:
With #15 merged, users are no longer advised / required to run their nodes as `root`.

closed 2019-05-08 by gavanderhoorn