Kawasaki-Robotics/khi_robot #24: issue using driver

yuokamoto, 2019-08-13:
We recently got the rs080 robot and trying to use the ros driver but we get this error

auxapi_login error ret=-4110
asapi_connect error ret=-4099
[ERROR] [1565690224.147868548]: [KhiRobotKrnxDriver] krnx_Open returned -0x1003
[ INFO] [1565690224.147958950]: [KhiRobotKrnxDriver] State 0: CONNECTING -> INIT
[ERROR] [1565690224.148058860]: Failed to open KHI robot

could you please help us resolve this issue?

yuokamoto, 2019-08-13:
Software version is ASE_010100X4K
Is this the cause of error? If so, how can I update?

yuokamoto, 2019-08-16:
This issue was solved by kind help from Kawasaki.
Thanks a lot.

closed 2019-08-16 by yuokamoto

Njima1572, 2021-03-10:
Got exactly same error. Is there any way to solve this on the user-end?

riyadh144, 2022-04-14:
Did anyone figure this out?
Please people when you find you find out a solution please tell the world

Landon-Getting, 2023-04-28:
I also encountered this issue - is there a solution? @matsui-hiro @y-yosuke

matsui-hiro, 2023-05-08:
Hi @Landon-Getting,

The AS system software version which you use may not be [a supported version](https://github.com/Kawasaki-Robotics/khi_robot/blob/master/docs/ConnectingRealRobot.md#12-supported-as-version).
Please check the version by Auxiliary function 0804 or Monitor command "ID".

Landon-Getting, 2023-05-15:
As @matsui-hiro suggested, the issue was that the AS system software on the controller was outdated. Kawasaki provided compatible version files and we used a USB to install them on the controller via the teach pendant.