Kawasaki-Robotics/khi_robot #25: [Feature Request] Add Z series support

mjshiggins, 2019-08-16:
Would it be possible to add support for the Z series armatures (I'm specifically looking for a driver for the ZX130L)? Alternatively, I'd be happy to write the driver myself. It would be helpful to add a page to `docs/` to assist users in adding new robots.

d-nakamichi, 2019-08-28:
Basically, KHI is going to add support for only new released robots from RS007. 

It would be helpful to submit a PR about adding a robot with following other `**_description` and `**_moveit_config`.
But KHI couldn't support it in real robot because it needs communication check and some tuning about ROS parameter for real robot.

mjshiggins, 2019-09-18:
Would you be open for me to add the comm check support to this repository?

d-nakamichi, 2019-09-20:
Yes, sure. It's very helpful.