Kawasaki-Robotics/khi_robot #39: Did I need do some jobs on the robot controller?

akjay, 2020-03-03:
I want to connect a real rs007n robot controller,which is ASF_010000010.Follow the ConnectingRealRobot.md file,I turned the robot to repeat mode and power on motor,change my Ubuntu 16.04 x64 to real_time mode.
When I launched rs007n_bringup.launch file with parameter viewer is true,the console output some log like this:
```
[ INFO] [1583224708.438422949]: Viewer mode
[ INFO] [1583224708.447050468]: KHI robot control started. [REALTIME]
[ INFO] [1583224708.867928302]: [KhiRobotKrnxDriver] KRNX Linux-x86_64 DEV 2.0.1 2019/11/07 13:20
[ INFO] [1583224708.867989845]: [KhiRobotKrnxDriver] State 0: INIT -> CONNECTING
[ INFO] [1583224708.868024180]: [KhiRobotKrnxDriver] Connecting to real controller: 192.168.0.2
No port_*.tbl. KRNX uses default port setting.
[WARN] [1583224709.048122]: DEPRECATION warning: --shutdown-timeout has no effect.
[INFO] [1583224709.048801]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1583224709.052085]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1583224709.055145]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1583224709.059882]: Loading controller: joint_state_controller
Loaded rs007n_joint_group_controller
[INFO] [1583224709.102875]: Loading controller: rs007n_arm_controller
[INFO] [1583224709.218577]: Controller Spawner: Loaded controllers: joint_state_controller, rs007n_arm_controller
[joint_group_controller_manager-6] process has finished cleanly
log file: /home/zhangh/.ros/log/566442f8-5d2a-11ea-bcff-f48e38f78bd2/joint_group_controller_manager-6*.log
[ INFO] [1583224711.353299533]: [KhiRobotKrnxDriver] State 0: CONNECTING -> INACTIVE
[ INFO] [1583224711.353588993]: [KhiRobotCommandService] Start
[INFO] [1583224711.356683]: Started controllers: joint_state_controller, rs007n_arm_controller
```
It seems that the joint value does not been uploaded,so the console not show joint value of the real robot,the Teach pendant shows no program running,so I wander did I need to do some jobs on the robot controller such as download some program to deal with the communication with ROS?
When I pressed CTRL+C to terminate the real_time control,the console print out this:
```
[controller_spawner-5] killing on exit
[robot_state_publisher-4] killing on exit
[INFO] [1583225044.185719]: Shutting down spawner. Stopping and unloading controllers...
[world_link_broadcaster-3] killing on exit
[INFO] [1583225044.186662]: Stopping all controllers...
[khi_robot_control-2] killing on exit
[ WARN] [1583225044.190860970]: received signal 2
[ INFO] [1583225044.191623272]: [KhiRobotKrnxDriver] State 0: INACTIVE -> DEACTIVATING
[ INFO] [1583225044.690389259]: [KhiRobotKrnxDriver] State 0: DEACTIVATING -> INACTIVE
[ INFO] [1583225044.831035009]: [KhiRobotKrnxDriver] State 0: INACTIVE -> DISCONNECTING
[ INFO] [1583225044.843570098]: [KhiRobotKrnxDriver] State 0: DISCONNECTING -> DISCONNECTED
[WARN] [1583225045.449121]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

```

akjay, 2020-03-04:
I have checked my Teach pendant, there is a program named "rb_rtc1" download automaticlly, so I changed it to run mode, but it shows that  "rtchome1 is not defined in line 2" when I pressed cycle start.While when I launched the moveit! node,the robot shown in rviz can synchronously with my real robot when I changed the robot to teach mode and move it manuelly.I try to subscribe the topic named "/jonit_states", it can show the real  robot's joints value correctlly, so I think the commucation worked well,but since when I run the program on Teach pendant, it will show that parameter not defined error,  but nothing shows on my PC's console,so what should I do on my robot controller,some files missed or the firmware did not satisfied?

akjay, 2020-03-04:
> I have checked my Teach pendant, there is a program named "rb_rtc1" download automaticlly, so I changed it to run mode, but it shows that "rtchome1 is not defined in line 2" when I pressed cycle [...]

Forgot to say,this time launched  in control mode, not view mode.

akjay, 2020-03-04:
I checked my Teach pendant, there is a program named "rb_rtc1" has been download automaticly,so I think the communication worked well in this situation.So I launched moveit! node(by running "roslaunch khi_rs007n_moveit_config moveit_planning_execution.launch"), I found that the robot model shown in rviz has the same position and attitude with my real robot, so I guess the joint value has been upload correctly.I subscribe the topic "/joint_state" see that the joint value is the same with my Teach pendant.
Then I changed to control mode(with viewer:=false by default),this time error occurs:
```
[ INFO] [1583305808.383662448]: KHI robot control started. [REALTIME]
process[controller_spawner-5]: started with pid [8109]
process[joint_group_controller_manager-6]: started with pid [8123]
[ INFO] [1583305808.494307428]: [KhiRobotKrnxDriver] KRNX Linux-x86_64 DEV 2.0.1 2019/11/07 13:20
[ INFO] [1583305808.494350278]: [KhiRobotKrnxDriver] State 0: INIT -> CONNECTING
[ INFO] [1583305808.494370971]: [KhiRobotKrnxDriver] Connecting to real controller: 192.168.0.2
No port_*.tbl. KRNX uses default port setting.
[WARN] [1583305808.714258]: DEPRECATION warning: --shutdown-timeout has no effect.
[INFO] [1583305808.714570]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1583305808.716487]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1583305808.719151]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1583305808.721039]: Loading controller: joint_state_controller
[INFO] [1583305808.734750]: Loading controller: rs007n_arm_controller
Loaded rs007n_joint_group_controller
[INFO] [1583305808.836326]: Controller Spawner: Loaded controllers: joint_state_controller, rs007n_arm_controller
[joint_group_controller_manager-6] process has finished cleanly
log file: /home/zhangh/.ros/log/2da45232-5de7-11ea-bcff-f48e38f78bd2/joint_group_controller_manager-6*.log
[ INFO] [1583305811.029522511]: [KhiRobotKrnxDriver] State 0: CONNECTING -> INACTIVE
[ INFO] [1583305811.029803247]: [KhiRobotKrnxDriver] State 0: INACTIVE -> ACTIVATING
[ INFO] [1583305811.029876499]: [KhiRobotCommandService] Start
[ERROR] [1583305816.637036875]: [KhiRobotKrnxDriver] Failed to activate: timeout
[ INFO] [1583305816.637110439]: [KhiRobotKrnxDriver] State 0: ACTIVATING -> ERROR
[ERROR] [1583305816.637161457]: Failed to activate KHI robot
[ INFO] [1583305816.639292384]: [KhiRobotKrnxDriver] State 0: ERROR -> DEACTIVATING
[ INFO] [1583305817.163849477]: [KhiRobotKrnxDriver] State 0: DEACTIVATING -> INACTIVE
[ INFO] [1583305817.264391555]: [KhiRobotKrnxDriver] State 0: INACTIVE -> DISCONNECTING
[ INFO] [1583305817.276822386]: [KhiRobotKrnxDriver] State 0: DISCONNECTING -> DISCONNECTED
```
It seems that timeout when activate the robot,I check the source code, but do not know why it comes.The user guide say to check  AS's robot program,but I really don't know what's wrong with the robot program,did the controller's firmware did not satisfied this?

akjay, 2020-03-04:
I have solved this,change mode on Teach pendant.Now the robot can move now.

closed 2020-07-04 by akjay

Ilya-Smolin, 2022-02-22:
Hello, i want to ask you in what mode was ethernet port in robot settings. And maybe u have faced this problem:

```INFO] [1645534431.976326167]: [KhiRobotKrnxDriver] State 0: INIT -> CONNECTING
[ INFO] [1645534431.976554345]: [KhiRobotKrnxDriver] Connecting to real controller: [192.168.1.2]
No port_*.tbl. KRNX uses default port setting.
allapi_open error ret=-8193
[ERROR] [1645534431.980037438]: [KhiRobotKrnxDriver] krnx_Open returned -0x2001
[ INFO] [1645534431.980543384]: [KhiRobotKrnxDriver] State 0: CONNECTING -> INIT
[ERROR] [1645534431.981306564]: Failed to open KHI robot
[ INFO] [1645534431.981946260]: [KhiRobotKrnxDriver] State 0: INIT -> DISCONNECTING
[ERROR] [1645534431.988772480]: [KhiRobotKrnxDriver] krnx_Close returned -0x2000
[ INFO] [1645534431.993668785]: [KhiRobotKrnxDriver] destructor
[ERROR] [1645534431.994257705]: [KhiRobotKrnxDriver] krnx_Close returned -0x2000
[WARN] [1645534433.132497]: DEPRECATION warning: --shutdown-timeout has no effect.
[INFO] [1645534433.135853]: Controller Spawner: Waiting for service controller_manager/load_controller
[WARN] [1645534433.142580]: wait_for_service(/controller_manager/load_controller): failed to contact, will keep trying
[khi_robot_control-2] process has died [pid 11583, exit code -11, cmd /home/ilya/catkin_ws/devel/lib/khi_robot_control/main --interface [192.168.1.2] --period 2 --robot RS007L __name:=khi_robot_control __log:=/home/ilya/.ros/log/797e299c-93de-11ec-9149-f14d68d93549/khi_robot_control-2.log].
log file: /home/ilya/.ros/log/797e299c-93de-11ec-9149-f14d68d93549/khi_robot_control-2*.log
```

akjay, 2022-03-02:
> Hello, i want to ask you in what mode was ethernet port in robot settings. And maybe u have faced this problem:
> 
> ```
> [ INFO] [1645534431.976554345]: [KhiRobotKrnxDriver] Connecting to real controller: [192.168.1.2]
> No port_*.tbl. KRNX uses default port setting.
> allapi_open error ret=-8193
> [ERROR] [1645534431.980037438]: [KhiRobotKrnxDriver] krnx_Open returned -0x2001
> [ INFO] [1645534431.980543384]: [KhiRobotKrnxDriver] State 0: CONNECTING -> INIT
> [ERROR] [1645534431.981306564]: Failed to open KHI robot
> [ INFO] [1645534431.981946260]: [KhiRobotKrnxDriver] State 0: INIT -> DISCONNECTING
> [ERROR] [1645534431.988772480]: [KhiRobotKrnxDriver] krnx_Close returned -0x2000
> [ INFO] [1645534431.993668785]: [KhiRobotKrnxDriver] destructor
> [ERROR] [1645534431.994257705]: [KhiRobotKrnxDriver] krnx_Close returned -0x2000
> [WARN] [1645534433.132497]: DEPRECATION warning: --shutdown-timeout has no effect.
> [INFO] [1645534433.135853]: Controller Spawner: Waiting for service controller_manager/load_controller
> [WARN] [1645534433.142580]: wait_for_service(/controller_manager/load_controller): failed to contact, will keep trying
> [khi_robot_control-2] process has died [pid 11583, exit code -11, cmd /home/ilya/catkin_ws/devel/lib/khi_robot_control/main --interface [192.168.1.2] --period 2 --robot RS007L __name:=khi_robot_control __log:=/home/ilya/.ros/log/797e299c-93de-11ec-9149-f14d68d93549/khi_robot_control-2.log].
> log file: /home/ilya/.ros/log/797e299c-93de-11ec-9149-f14d68d93549/khi_robot_control-2*.log
> ```

Nothing need to do with the robot,just connect the EtherNet cable,the driver will use port 23 I think,it is the talnet port.