Kawasaki-Robotics/khi_robot #40: RTC SWITCH turned OFF 0: ano:1

frash83, 2020-05-14:
How can I debug this issue?

RTC SWITCH turned OFF 0: ano:1

It happens 9 out of 10 times!

d-nakamichi, 2020-05-15:
Robot controller executes a robot program using RTC_SWITCH during ROS control.
This robot program may be holded due to something (error, external hold, etc.).
Could you look at your teach pendant and check the status of robot program named "rb_rtc1"?

frash83, 2020-05-15:
I mean it is fine exactly the same as RTC_SWITCH turned off! 
RUN > MOTOR > CYCLE [green] and I can see rb_rtc1 as well 
I think after CYCLE, I should not UpdateState very fast ... I should wait a little for RTC_SWITCH to sync? Is that right?

d-nakamichi, 2020-05-18:
> I think after CYCLE, I should not UpdateState very fast ... I should wait a little for RTC_SWITCH to sync? Is that right?

Driver waits a RTC_SWITCH's ON when ACTIVATING state is run, so it may be not a problem...
Could you show the entire terminal output?

frash83, 2020-05-18:
This is a sample output:

[2020-05-13 16:51:51.366] [info] [khi_driver.h:201] [KhiKrnxDriver] KRNX Linux64 DEV 008 2019/02/04 17:43
[2020-05-13 16:51:51.366] [info] [khi_driver.h:201] [KhiKrnxDriver] Connecting to real controller: 10.10.1.111
[2020-05-13 16:51:53.004] [info] [khi_driver.h:169] [KhiKrnxDriver] State CHANGE 0: NOT_CONNECTED -> CONNECTED
[2020-05-13 16:51:54.085] [info] [khi_driver.h:169] [KhiKrnxDriver] State CHANGE 0: CONNECTED -> ACTIVATING
[2020-05-13 16:51:55.402] [info] [khi_driver.h:169] [KhiKrnxDriver] State CHANGE 0: ACTIVATING -> ACTIVE
[2020-05-13 16:51:55.404] [error] [khi_driver.h:211] [KhiKrnxDriver] RTC SWITCH turned OFF 0: ano:1
[2020-05-13 16:51:55.409] [info] [khi_driver.h:169] [KhiKrnxDriver] State CHANGE 0: ACTIVE -> CONNECTED
[2020-05-13 16:51:55.473] [error] [khi_driver.cpp:261] An error has occured, to resume control of the robot:

d-nakamichi, 2020-05-18:
> [2020-05-13 16:51:51.366] [info] [khi_driver.h:201] [KhiKrnxDriver] KRNX Linux64 DEV 008 2019/02/04 17:43

Your driver is old alpha test version (not released in this GitHub repository) and may have some ploblems.
Could you try our latest version (master) instead of old one?

frash83, 2020-08-18:
By adding some delay (time.sleep(x)) between activation and rtc switch, it has been solved.