Kawasaki-Robotics/khi_robot #45: add URDF description for RS003N

k-okada, 2020-10-08:
@d-nakamichi hi, how can we add urdf model for RS003 ?
https://robotics.kawasaki.com/ja1/products/CAD-disclaimer/ said "上述の利用目的以外にCADデータを複製、公衆送信、改変、転用、販売などの利用をすることは、事前に当社から許可を得ない限り禁止します。" So how can we get approval?

cc: @n-ando

d-nakamichi, 2020-10-13:
You can use our CAD data for private ROS simulation.
But currently we don't approve merging the new robot arm package to this repository.
This is because it needs our some technical parameter tuning to use this in real robot.
We are now internally discussing it to enable you to develop the new robot arm packages in real robot and merge it to this repository.

To add urdf model,

1. convert CAD STEP to STL
2. create a urdf package following other `khi_robot` arm packages and arm parameters
3. create moveit/gazebo package following other `khi_robot` moveit packages

k-okada, 2020-10-13:
Hello Nakamich-san

Thanks for the answer. I understand the situation.

We're planning to start a new Robot Programming Course with RS003N under
NEDO support, which is much safer than using RS007.
1) To use public courses, I think putting RS003N model is very important.
As for now, we only interested in simulation. So putting the existing model
with "experimental only for simulation" would be ok for us
2) Or I think we could help you to setup parameter tuning, if we need to
support a real robot if we want to put the robot model in public.
--
◉ Kei Okada