Kawasaki-Robotics/khi_robot #47: MoveIt Robot motion

elizatscott, 2020-12-16:
While working with the real robot I am not getting a sequential motion like
home > position1 > position2 > position3....
Instead I am getting motions like
home > poistion1 > home > position2 > home > poisition3....

When checked with gazebo simulation it works correctly whereas when I am working with the real robot I am getting the error mentioned above.

d-nakamichi, 2020-12-17:
Could you tell me the AS system software version which you use.
You can check the version by Auxiliary function 0804  or Monitor command "ID".

elizatscott, 2021-01-05:
[image]

d-nakamichi, 2021-01-06:
To use our driver in F series, the robot controller's AS system should be higher than `ASF_01000000W`. 
([docs/ConnectingRealRobot.md](docs/ConnectingRealRobot.md)) 
Could you ask your KHI partner to upgrade AS system?

elizatscott, 2021-01-11:
Still I could not get the required target motion.
Instead I am getting motions like
home > poistion1 > home > position2 > home > poisition3.... This is the motion that is happening still.
These are my code snippets. I have connected the robot to the ubuntu through the vmware workstation. Does it have anything to deal with this ?
[image]
[image]

d-nakamichi, 2021-01-12:
About,

>home > position1 > home > position2 > home > poisition3....

If,
- you set `set_max_velocity_scaling_factor(0.1)`
- you set a home position which is not equal to [JT1:0 JT2:0 JT3:0 JT4:0 JT5:0 JT6:0]
- the arm moves `home(0) > position1 > home(1) > position2 > home(2) > poisition3...`,

How is the motion `position1 > home(1)` like?

- Is the motion speed same as the speed : `velocity_scaling(0.1)`?
- Is home(1) same as `home(0)`, or  [JT1~6: 0]?

elizatscott, 2021-01-15:
Yes! The motion speed is same as velocity scaling but it is maintained only during the motion of home > poisition 1 and not during poisition 1 > home.
The home 1 is not same as home 0 but it is going near home 0

d-nakamichi, 2021-01-17:
It seems to be a problem on `khi_robot` side.
To reproduce your problem, let me know your system environment in detail.

- Ubuntu/ROS version
- Current AS system
- `khi_robot` version or revision
- Motion script: Is it same as https://github.com/Kawasaki-Robotics/khi_robot/issues/47#issuecomment-757865376 ?

elizatscott, 2021-01-28:
I am using VMware for Ubuntu 16.04
ASF_010000019
THE SAME KHI ROBOT VERSION CURRENTLY IN THE GITHUB AND THE SAME MOTION SCRIPT @d-nakamichi