Kawasaki-Robotics/khi_robot #49: request for melodic release

k-okada, 2021-01-14:
@d-nakamichi @matsui-hiro Could you release khi_robot to melodic? We have confiremd it is working on melodic as it is with @sktometometo

Cc: @n-ando

d-nakamichi, 2021-01-17:
We also confirmed it with melodic.
OK, we'll make a plan to release the current revision to Melodic/Kinetic distro and do it by this month.

k-okada, 2021-01-18:
@d-nakamichi Thakn you!

d-nakamichi, 2021-02-25:
Now, I've released v1.2.0 for Kinetic/Melodic
https://wiki.ros.org/khi_robot

k-okada, 2021-02-25:
thank you!
--
◉ Kei Okada

d-nakamichi, 2021-02-26:
If there is no additional request, could you close this issue?

k-okada, 2021-02-26:
hoep we could have rs003 in melodic https://github.com/Kawasaki-Robotics/khi_robot/issues/45, but closeing this issue for now.

closed 2021-02-26 by k-okada