Kawasaki-Robotics/khi_robot #51: RTC SWITCH turned OFF 0: ano:1

r-ym, 2021-07-19:
Hello,

I have been trying to use MoveIt with my rs007l but every time I run 
`roslaunch khi_robot_bringup rs007l_bringup.launch ip:=<IP address>` I get an error message, which I have logged below with terminal output. This may be similar to the earlier issue https://github.com/Kawasaki-Robotics/khi_robot/issues/40

<details>
  <summary>Error Log</summary>

```` 
```
 $roslaunch khi_robot_bringup rs007l_bringup.launch
... logging to /root/.ros/log/bd82e2c4-e860-11eb-bfe1-0242ac110002/roslaunch-d24dc7cf03df-18862.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://d24dc7cf03df:39637/

SUMMARY
========

PARAMETERS
 * /RS007L/joint_limits/joint1/has_acceleration_limits: True
 * /RS007L/joint_limits/joint1/has_position_limits: True
 * /RS007L/joint_limits/joint1/has_velocity_limits: True
 * /RS007L/joint_limits/joint1/max_acceleration: 5.65050345333
 * /RS007L/joint_limits/joint1/max_position: 1.57
 * /RS007L/joint_limits/joint1/max_velocity: 6.4577
 * /RS007L/joint_limits/joint1/min_position: -1.57
 * /RS007L/joint_limits/joint2/has_acceleration_limits: True
 * /RS007L/joint_limits/joint2/has_position_limits: True
 * /RS007L/joint_limits/joint2/has_velocity_limits: True
 * /RS007L/joint_limits/joint2/max_acceleration: 1.80350689372
 * /RS007L/joint_limits/joint2/max_position: 2.35619449019
 * /RS007L/joint_limits/joint2/max_velocity: 5.4105
 * /RS007L/joint_limits/joint2/min_position: -2.35619449019
 * /RS007L/joint_limits/joint3/has_acceleration_limits: True
 * /RS007L/joint_limits/joint3/has_position_limits: True
 * /RS007L/joint_limits/joint3/has_velocity_limits: True
 * /RS007L/joint_limits/joint3/max_acceleration: 3.57792496658
 * /RS007L/joint_limits/joint3/max_position: 3.49065850399
 * /RS007L/joint_limits/joint3/max_velocity: 7.1558
 * /RS007L/joint_limits/joint3/min_position: -3.49065850399
 * /RS007L/joint_limits/joint4/has_acceleration_limits: True
 * /RS007L/joint_limits/joint4/has_position_limits: True
 * /RS007L/joint_limits/joint4/has_velocity_limits: True
 * /RS007L/joint_limits/joint4/max_acceleration: 23.99827721492
 * /RS007L/joint_limits/joint4/max_position: 0.785
 * /RS007L/joint_limits/joint4/max_velocity: 9.5992
 * /RS007L/joint_limits/joint4/min_position: -0.785
 * /RS007L/joint_limits/joint5/has_acceleration_limits: True
 * /RS007L/joint_limits/joint5/has_position_limits: True
 * /RS007L/joint_limits/joint5/has_velocity_limits: True
 * /RS007L/joint_limits/joint5/max_acceleration: 23.99827721492
 * /RS007L/joint_limits/joint5/max_position: 0
 * /RS007L/joint_limits/joint5/max_velocity: 9.5992
 * /RS007L/joint_limits/joint5/min_position: -2.18166156499
 * /RS007L/joint_limits/joint6/has_acceleration_limits: True
 * /RS007L/joint_limits/joint6/has_position_limits: True
 * /RS007L/joint_limits/joint6/has_velocity_limits: True
 * /RS007L/joint_limits/joint6/max_acceleration: 13.08996938995
 * /RS007L/joint_limits/joint6/max_position: 0.785
 * /RS007L/joint_limits/joint6/max_velocity: 17.4532
 * /RS007L/joint_limits/joint6/min_position: -0.785
 * /cartesian_compliance_controller/compliance_ref_link: tool0
 * /cartesian_compliance_controller/end_effector_link: tool0
 * /cartesian_compliance_controller/ft_sensor_ref_link: sensor_link_tip
 * /cartesian_compliance_controller/joints: ['joint1', 'joint...
 * /cartesian_compliance_controller/pd_gains/rot_x/p: 0.01
[... 124 lines cut ...]
[ INFO] [1626678779.100600300]: [KhiRobotKrnxDriver] State 0: INACTIVE -> ACTIVATING
[ INFO] [1626678779.100615878]: [KhiRobotCommandService] Start
[ INFO] [1626678780.358965439]: [KhiRobotKrnxDriver] State 0: ACTIVATING -> ACTIVE
[INFO] [1626678780.361419]: Started controllers: joint_state_controller, rs007l_arm_controller
[ERROR] [1626678800.361351228]: [KhiRobotKrnxDriver] krnx_PrimeRtcCompData returned -0x100A
[ INFO] [1626678800.361454328]: [KhiRobotKrnxDriver] State 0: ACTIVE -> ERROR
[ WARN] [1626678800.361552031]: JT1:-0.095805,-0.095805,0.000000,0.000000,-0.095805,-0.095805
[ WARN] [1626678800.361608894]: JT1:-5.489209,-5.489209,0.000000,0.000000,-5.489209,-5.489209
[ WARN] [1626678800.361715906]: JT2:0.625314,0.625314,0.000000,-0.000070,0.625385,0.625302
[ WARN] [1626678800.361786377]: JT2:35.827880,35.827880,0.000000,-0.004030,35.831910,35.827148
[ WARN] [1626678800.361850002]: JT3:-1.059591,-1.059591,0.000000,0.000132,-1.059723,-1.059310
[ WARN] [1626678800.361916102]: JT3:-60.710082,-60.710082,0.000000,0.007588,-60.717670,-60.694015
[ WARN] [1626678800.361979661]: JT4:-0.053183,-0.053183,0.000000,0.000000,-0.053183,-0.053183
[ WARN] [1626678800.362042381]: JT4:-3.047168,-3.047168,0.000000,0.000000,-3.047168,-3.047168
[ WARN] [1626678800.362111643]: JT5:-0.041945,-0.041945,0.000000,0.000072,-0.042017,-0.041945
[ WARN] [1626678800.362168844]: JT5:-2.403259,-2.403259,0.000000,0.004120,-2.407379,-2.403259
[ WARN] [1626678800.362238197]: JT6:-0.941506,-0.941506,0.000000,0.034908,-0.976414,-0.976412
[ WARN] [1626678800.362298518]: JT6:-53.944318,-53.944318,0.000000,2.000071,-55.944389,-55.944308
[ERROR] [1626678800.362345013]: [KhiRobotKrnxDriver] [krnx_PrimeRtcCompData] ano:1 [jt]pos:vel:status [1]-0.0958:0.0000:0 [2]0.6253:-0.0352:0 [3]-1.0593:0.0662:0 [4]-0.0532:0.0000:0 [5]-0.0419:0.0360:0 [6]-0.9764:17.4539:4
[ERROR] [1626678840.470874782]: [KhiRobotKrnxDriver] RTC SWITCH turned OFF 0: ano:1
[ INFO] [1626678840.470904725]: [KhiRobotKrnxDriver] State 0: ERROR -> DEACTIVATING
[ INFO] [1626678840.982626963]: [KhiRobotKrnxDriver] State 0: DEACTIVATING -> INACTIVE
```
````
</details>

d-nakamichi, 2021-07-28:
Please follow [docs/ConnectingRealRobot.md#3-1-start-the-real-time-control](https://github.com/Kawasaki-Robotics/khi_robot/blob/master/docs/ConnectingRealRobot.md#3-1-start-the-real-time-control).

After connecting your PC and Robot Controller by Ethernet and putting them into same network, please execute bringup by `roslaunch khi_robot_bringup rs007n_bringup.launch ip:=[Controller’s IP address]`

r-ym, 2021-07-30:
Hello, I did put in the correct IP but turns out the machine at work is the "L" variant and not the "N". Switching the bringups solved the issue. Very sorry for the inconsequential issue. Will be closing it. Thank you very much!

closed 2021-07-30 by r-ym