Kawasaki-Robotics/khi_robot #63: what(): Duration is out of dual 32-bit range
(And it isn't writing the log files)

vineet131, 2022-05-03:
I'm running multiple nodes on ROS Noetic, and khi_robot_control is one of them. It's working fine for a few runs, but after a random while, it displays the following error:

```
terminate called after throwing an instance of 'std::runtime_error'
what(): Duration is out of dual 32-bit range
```

This has been difficult to debug since the log file for khi_robot_control is not being written to disk (all my other nodes are writing the log files to disk just fine)

MasaoAramoto, 2022-05-03:
2022/05/09 まで不在です。

4/18-4/23は出張のため不在です。

I'm not in my office during 4/29-5/8.
Apologies for the delay in responding.

メモ: これは次のメールへの自動応答です。  "[Kawasaki-Robotics/khi_robot]
what(): Duration is out of dual 32-bit range=0A(And it isn't writing the
log files) (Issue #63)" (送信日時: 03/05/2022 14:39:14) .

このユーザーの不在期間中に、あなたが受け取る不在通知はこの一通のみです。

matsui-hiro, 2022-05-13:
Could you send us more terminal log information of `khi_robot_control` ?
That might help our investigation.

vineet131, 2022-05-16:
That's what I'm trying to say, the log file for this node isn't getting written to the .ros folder even though all other nodes are writing to the folder just fine.

Khi_robot_controlのlog fileは保存されていません。

matsui-hiro, 2022-05-16:
OK. 
I understood the main issue in this ticket is `Khi_robot_control `did not write the log file to disk.

vineet131, 2022-05-18:
Nope. I clearly wrote the issue in the first comment. Please refer to that.

I am unable to provide the log because it isn't being written to disk, I don't know why.

matsui-hiro, 2022-05-20:
> Nope. I clearly wrote the issue in the first comment. Please refer to that.

OK.

> I am unable to provide the log because it isn't being written to disk, I don't know why.

Is there any output other than error message?
If you could know the the situation that occurring error, please tell me.
Robot type, khi_robot_control  state,  etc.

vineet131, 2022-06-17:
I did not get this error again. Maybe PR #65 solved it for me

closed 2022-06-17 by vineet131