Kawasaki-Robotics/khi_robot #66: connecting real robot

yamamoto59, 2022-06-16:
Hi.
I tried to connect real duaro robot.But ROS says 
"allapi_open error ret=8192"
I searched for what is this erroer.But i can't find the solution.
Please tell me what is this error and what should i do.

matsui-hiro, 2022-06-17:
Error code "8192" is "Socket Create Error (0x2000)".
Could you check for connectivity by ping?

I would like to know more detailed situation.
What is the software version of the controller?
Please send the log of `duaro_bringup.launch.`

yamamoto59, 2022-06-22:
I tryed to ping . It succeeded from ROS PC.
The software version is ADU00003.

```duaro_bringup.launch``` displayed the error log as 
```
[INFO] [1655856325.866843]: Loading controller: duaro_upper_arm_controller
[INFO] [1655856325.900420]: Controller Spawner: Loaded controllers: joint_state_controller, duaro_lower_arm_controller, duaro_upper_arm_controller
[joint_group_controller_manager-6] process has finished cleanly
log file: /home/mirai/.ros/log/039746e8-f1bf-11ec-b948-9cda3ed9e9b4/joint_group_controller_manager-6*.log
auxapi_login error ret=-8200
asapi_connect error ret=-4099
[ERROR] [1655856328.954040448]: [KhiRobotKrnxDriver] krnx_Open returned -0x1003
[ INFO] [1655856328.954081543]: [KhiRobotKrnxDriver] State 0: CONNECTING -> INIT
[ERROR] [1655856328.954182725]: Failed to open KHI robot
[ INFO] [1655856328.954219471]: [KhiRobotKrnxDriver] State 0: INIT -> DISCONNECTING
[ERROR] [1655856328.954297993]: [KhiRobotKrnxDriver] krnx_Close returned -0x2000
[ INFO] [1655856328.954347290]: [KhiRobotKrnxDriver] destructor
[ERROR] [1655856328.954389753]: [KhiRobotKrnxDriver] krnx_Close returned -0x2000
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 199, in main
    resp = switch_controller(loaded, [], 2)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1655856329.071108]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1655856329.071503]: Stopping all controllers...
[WARN] [1655856329.074006]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
[khi_robot_control-2] process has died [pid 4070, exit code -11, cmd /home/mirai/duaro_ws/devel/lib/khi_robot_control/main --interface 192.168.0.2 --period 4 --robot WD002N __name:=khi_robot_control __log:=/home/mirai/.ros/log/039746e8-f1bf-11ec-b948-9cda3ed9e9b4/khi_robot_control-2.log].
log file: /home/mirai/.ros/log/039746e8-f1bf-11ec-b948-9cda3ed9e9b4/khi_robot_control-2*.log
[controller_spawner-5] process has died [pid 4080, exit code 1, cmd /opt/ros/kinetic/lib/controller_manager/spawner --shutdown-timeout 0.1 joint_state_controller duaro_lower_arm_controller duaro_upper_arm_controller __name:=controller_spawner __log:=/home/mirai/.ros/log/039746e8-f1bf-11ec-b948-9cda3ed9e9b4/controller_spawner-5.log].
log file: /home/mirai/.ros/log/039746e8-f1bf-11ec-b948-9cda3ed9e9b4/controller_spawner-5*.log
```
The connection with the controller seems to be failing.
AS software and the real robot are connected on the same network.

yamamoto59, 2022-06-22:
I noticed error log was changed after posting.
```allapi_open error ret=8192```
This may change from time to time.
```auxapi_login error ret=-8200```
```asapi_connect error ret=-4099```

matsui-hiro, 2022-06-22:
I think you sue D series controller.
For duAro, ROS can only be used by F series controller .
(https://github.com/Kawasaki-Robotics/khi_robot/blob/master/docs/ConnectingRealRobot.md#f-series.)

yamamoto59, 2022-06-22:
I made a simple mistake.
Is it completely impossible to get it to work with a D controller?
AS system software version is AS_0920004.

matsui-hiro, 2022-06-22:
AS_09200004 is the soft version of D controller.

> Is it completely impossible to get it to work with a D controller?

Yes. If you use ROS with duAro, please use F series controller.