Kawasaki-Robotics/khi_robot #73: How to make the robot active again after sending HOLD with khi_robot_command_service

wakasaki41, 2022-09-23:
We are working with RS080n and tried to use khi_robot_command_service to send HOLD and CONTINUE signals from the safety system.
The HOLD signal worked well, but we could not get it back to the active state with the CONTINUE signal.

How can I transition the robot to the active state?

matsui-hiro, 2022-09-27:
When stopping the robot,  you can use `type : as`  `cmd: HOLD` of khi_robot_command_service  .

But to reactivate driver, please use `type : driver` (not `type : as`)  .

> string type -> "driver"
> string cmd -> "restart"

Sending `cmd: restart`  activates driver.

Please see [this](https://github.com/Kawasaki-Robotics/khi_robot/blob/master/docs/ConnectingRealRobot.md#restart-driver-for-active) for details.

wakasaki41, 2022-09-30:
Thank you very much.
I have tried the reboot command as it is.
What kind of error is this?
```
$ rosservice call /khi_robot_command_service "type: 'driver'
cmd: 'restert'"
driver_ret: -4096
as_ret: 0
cmd_ret: "INVALID CMD"
```

matsui-hiro, 2022-09-30:
Hi @wakasaki41,

From the error log, I found a spelling mistake.
Please execute `cmd:restart` not `cmd:restert`.

wakasaki41, 2022-09-30:
Thank you very much , It worked ‼!

closed 2022-09-30 by wakasaki41