Kawasaki-Robotics/khi_robot #74: Communication with the robot controller is lost.

wakasaki41, 2022-09-23:
We are dealing with RS080n, we are using rs080n_bringup.launch to run the robot, but sometimes we lose communication with the robot. At that time, if we send a ping command, we get 100% packet loss.

The cases where this phenomenon is observed are when I press emergency stop to stop the robot or when I use Ctrl+C to exit rs080n_bringup.launch. However, this does not happen every time.

matsui-hiro, 2022-09-29:
Hi @wakasaki41,

I have tried Ctrl+C to exit rs080n_bringup.launch and emergency stop during communication but packet loss did not occur.

I would like to know more detailed situation.
What is the software version of the controller?
Please send the log of rs080n_bringup.launch if you have.

wakasaki41, 2022-09-30:
Where can I check the controller version?

matsui-hiro, 2022-09-30:
You can check the version by Auxiliary function 0804 or Monitor command "ID".

wakasaki41, 2022-09-30:
[image]

wakasaki41, 2022-09-30:
We thought there was a negotiation problem, so we disabled auto-negotiation and fixed the communication speed at 1000 Mbps.
We also changed to use khi_robot_command_service to send a quit command to the Driver to kill it.

So far the problem has not recurred, but if it happens again, we will contact you.

Thank you very much.