Kawasaki-Robotics/khi_robot #76: How to read status of I/O using khi_robot_command service message

Ohaginia, 2022-10-13:
How can I use the khi_robot_command service message to read the status of external and internal inputs and outputs?

I sent the following command, which returned an error message.

**_Terminal 1_**
> rosservice call /khi_robot_command_service "type: 'as' cmd: 'IO'"
(freeze...)

**_Terminal 2_**
> roslaunch khi_robot_bringup rs080n_bringup.launch
(omission)
NODES
  /
    controller_spawner (controller_manager/spawner)
    joint_group_controller_manager (controller_manager/controller_manager)
    khi_robot_control (khi_robot_control/main)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    world_link_broadcaster (tf/static_transform_publisher)
auto-starting new master
process[master]: started with pid [9708]
ROS_MASTER_URI=http://localhost:11311
setting /run_id to 2ade9bd4-4af7-11ed-a364-2046a1083a2b
process[rosout-1]: started with pid [9719]
started core service [/rosout]
process[khi_robot_control-2]: started with pid [9726]
process[world_link_broadcaster-3]: started with pid [9727]
process[robot_state_publisher-4]: started with pid [9728]
[ INFO] [1665666096.804324751]: KHI robot control started. [REALTIME]
process[controller_spawner-5]: started with pid [9744]
process[joint_group_controller_manager-6]: started with pid [9750]
[ INFO] [1665666096.820296365]: [KhiRobotKrnxDriver] KRNX Linux-x86_64 DEV 2.3.5 2022/05/09 20:23
[ INFO] [1665666096.820341026]: [KhiRobotKrnxDriver] State 0: INIT -> CONNECTING
[ INFO] [1665666096.820362085]: [KhiRobotKrnxDriver] Connecting to real controller: 192.168.0.2
No port_*.tbl. KRNX uses default port setting.
[WARN] [1665666097.168819]: DEPRECATION warning: --shutdown-timeout has no effect.
[INFO] [1665666097.170123]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1665666099.457898163]: [KhiRobotKrnxDriver] State 0: CONNECTING -> INACTIVE
[ INFO] [1665666099.458246222]: [KhiRobotCommandService] Start
[ INFO] [1665666099.491354672]: [KhiRobotKrnxDriver] State 0: INACTIVE -> ACTIVATING
[INFO] [1665666099.586963]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1665666099.589087]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1665666099.590803]: Loading controller: joint_state_controller
Loaded 'rs080n_joint_group_controller'
[INFO] [1665666099.596617]: Loading controller: rs080n_arm_controller
[INFO] [1665666099.617694]: Controller Spawner: Loaded controllers: joint_state_controller, rs080n_arm_controller
[joint_group_controller_manager-6] process has finished cleanly
log file: /home/owner/.ros/log/2ade9bd4-4af7-11ed-a364-2046a1083a2b/joint_group_controller_manager-6*.log
[ INFO] [1665666100.339286722]: [KhiRobotKrnxDriver] State 0: ACTIVATING -> ACTIVE
[INFO] [1665666100.341858]: Started controllers: joint_state_controller, rs080n_arm_controller
**[ERROR] [1665666104.267173269]: [KhiRobotKrnxDriver] krnx_ExecMon() returned -0x100E**

matsui-hiro, 2022-10-14:
Hi @Ohaginia ,

If you want to get the latest status of external input/output signal and internal signal, you can use `type : deiver cmd: "get_signal [NUM]"` of khi_robot_command_service .

> string type-> "driver"
> string cmd -> "get_signal [NUM]"

Please see [this](https://github.com/Kawasaki-Robotics/khi_robot/blob/master/docs/ConnectingRealRobot.md#get-signal-status-of-num) for details.

Ohaginia, 2022-10-25:
I have tried this and it works great.

```
rosservice call /khi_robot_command_service "type: 'driver'
cmd: 'get_signal [1]'"  
```

The result is as follows

```
driver_ret: -4096
as_ret: 0
cmd_ret: "INVALID ARGS"
```

What is wrong?

matsui-hiro, 2022-10-26:
"[" and "]"  are not needed.

Could you execute as follows?

```
rosservice call /khi_robot_command_service "type: 'driver'
cmd: 'get_signal 1'"  
```

Ohaginia, 2022-10-26:
Thank you very much.
set_signal worked fine, but get_signal did not.
```
rosservice call /khi_robot_command_service "type: 'driver'
cmd: 'set_signal 1'" 
driver_ret: 0
as_ret: 0
cmd_ret: ''
```

```
 rosservice call /khi_robot_command_service "type: 'driver'
cmd: 'get_signal 1'"
driver_ret: -4097
as_ret: 0
cmd_ret: ''
```

```
rosservice call /khi_robot_command_service "type: 'driver'
cmd: 'set_signal -1'"
driver_ret: 0
as_ret: 0
cmd_ret: ''
```

```
rosservice call /khi_robot_command_service "type: 'driver'
cmd: 'get_signal 1'"
driver_ret: -4097
as_ret: 0
cmd_ret: ''
```

What is wrong?

matsui-hiro, 2022-10-27:
Does the **"-4097"** error return after turning off/on the ROS-PC and the robot controller?

If it still happens , I would like to know the details.
What is the software version of the controller? 
(You can check the version by Auxiliary function 0804 or Monitor command "ID".)
And please let me know the terminal log of rs080n_bringup.launch when you get the **"-4097"** error.

Alternatively, you can also get the status of signal by `type : as cmd: "type sig(1)"` instead of `type : driver cmd: "get_signal 1"`.

Ohaginia, 2022-11-18:
It was confirmed that the same error occurs when the ROS-PC and controller power is turned on and off.

When I send "type : as cmd: "type sig(1)", I get a signal input instead of a signal state acquisition.

And after much thought, I succeeded in using this to get the state of the signal.
type : as cmd: "type PRINT(SIG(1))"

matsui-hiro, 2022-11-21:
Did you execute as follows?

```
rosservice call /khi_robot_command_service "type: 'as'
cmd: 'sig(1)'"
```

I think the following command can gets the status of output signal `1` correctly.

```
rosservice call /khi_robot_command_service "type: 'as'
cmd: 'type sig(1)'"
```
The following command works similarly.

```
rosservice call /khi_robot_command_service "type: 'as'
cmd: 'print sig(1)'"
```