Kawasaki-Robotics/khi_robot #78: RTC Switch Error When Having different ROS Master

siddharthbhurat4, 2023-03-18:
When my laptop is the ROS master, and when I run the duaro_bringup.launch, everything works fine. But when another robot is a ROS master and when I run duaro_bringup.launch, it give the RTC Error that is "krnx_PrimeRtcCompData returned -0x100A" and then in the end it says "RTC Switch turned OFF 0: ano:1".

Please can I know, how to solve this issue.

Thank you

matsui-hiro, 2023-03-22:
Does the Toyota HSR robot controller make plans and send to the duaro robot controller ? 
0x100A is command value error. The speed may have exceeded the upper limit.

Was there the error message like bellow ?

`[KhiRobotKrnxDriver] [krnx_PrimeRtcCompData] ano:[arm_no] [jt]pos:vel:status [JT1]%.4f:%.4f:%d ~ [JT6]%.4f:%.4f:%d`

IF so, please follow [docs/ConnectingRealRobot.md#6-error-and-troubleshooting](https://github.com/Kawasaki-Robotics/khi_robot/blob/master/docs/ConnectingRealRobot.md#6-error-and-troubleshooting). 
You can get more detail error cause from the "status" value.

siddharthbhurat4, 2023-03-22:
Yes, the error was like the which you have mentioned above. I am actually running just the bringup.launch file which just launches the joint controllers, the urdfs', so that I can visualize them in RViz. I am not running the moveit node, so I think there are no plans being sent to the duaro robot. And even though Toyota HSR is the master, all the launch files are being run from the same laptop which is now slave to the Toyota HSR.

So what I am currently trying to do is, I have my laptop which is slave to Toyota HSR (The ROS Master), Duaro is also the slave to Toyota HSR. And I am running all the launch files from laptop where they are actually installed. 

(Just putting this up again - Whenever the laptop is master, Duaro works perfectly fine, the moment I switch from laptop as master to Toyota HSR as the master, it gives that error of RTC switch (sometimes it works once in like 10-15 times of giving error), but I am still running everything from laptop itself, just that the master is changed)

And then it gives this error. 

<hsrb>~$ roslaunch khi_robot_bringup duaro_bringup.launch ip:=192.168.50.10
... logging to /home/bartonlab-user/.ros/log/20230322-164526_f274ad62-c8d0-11ed-a2b7-0013953132f2/roslaunch-me-bartonlab33-61923.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
 
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://192.168.50.202:38691/
 
SUMMARY
========
 
PARAMETERS
 * /WD002N/joint_limits/lower_joint1/has_acceleration_limits: True
 * /WD002N/joint_limits/lower_joint1/has_position_limits: True
 * /WD002N/joint_limits/lower_joint1/has_velocity_limits: True
 * /WD002N/joint_limits/lower_joint1/max_acceleration: 2.09439510239
 * /WD002N/joint_limits/lower_joint1/max_position: 2.96705972839
 * /WD002N/joint_limits/lower_joint1/max_velocity: 4.1887
 * /WD002N/joint_limits/lower_joint1/min_position: -2.96705972839
 * /WD002N/joint_limits/lower_joint2/has_acceleration_limits: True
 * /WD002N/joint_limits/lower_joint2/has_position_limits: True
 * /WD002N/joint_limits/lower_joint2/has_velocity_limits: True
 * /WD002N/joint_limits/lower_joint2/max_acceleration: 7.85398163397
 * /WD002N/joint_limits/lower_joint2/max_position: 2.44346095279
 * /WD002N/joint_limits/lower_joint2/max_velocity: 5.4977
 * /WD002N/joint_limits/lower_joint2/min_position: -2.44346095279
 * /WD002N/joint_limits/lower_joint3/has_acceleration_limits: True
 * /WD002N/joint_limits/lower_joint3/has_position_limits: True
 * /WD002N/joint_limits/lower_joint3/has_velocity_limits: True
 * /WD002N/joint_limits/lower_joint3/max_acceleration: 3.0
 * /WD002N/joint_limits/lower_joint3/max_position: 0.15
 * /WD002N/joint_limits/lower_joint3/max_velocity: 0.255
 * /WD002N/joint_limits/lower_joint3/min_position: 0
 * /WD002N/joint_limits/lower_joint4/has_acceleration_limits: True
 * /WD002N/joint_limits/lower_joint4/has_position_limits: True
 * /WD002N/joint_limits/lower_joint4/has_velocity_limits: True
 * /WD002N/joint_limits/lower_joint4/max_acceleration: 52.35987755983
 * /WD002N/joint_limits/lower_joint4/max_position: 6.28318530718
 * /WD002N/joint_limits/lower_joint4/max_velocity: 10.4717
 * /WD002N/joint_limits/lower_joint4/min_position: -6.28318530718
 * /WD002N/joint_limits/upper_joint1/has_acceleration_limits: True
 * /WD002N/joint_limits/upper_joint1/has_position_limits: True
 * /WD002N/joint_limits/upper_joint1/has_velocity_limits: True
 * /WD002N/joint_limits/upper_joint1/max_acceleration: 2.09439510239
 * /WD002N/joint_limits/upper_joint1/max_position: 8.72664625997
 * /WD002N/joint_limits/upper_joint1/max_velocity: 4.1887
 * /WD002N/joint_limits/upper_joint1/min_position: -2.44346095279
 * /WD002N/joint_limits/upper_joint2/has_acceleration_limits: True
 * /WD002N/joint_limits/upper_joint2/has_position_limits: True
 * /WD002N/joint_limits/upper_joint2/has_velocity_limits: True
 * /WD002N/joint_limits/upper_joint2/max_acceleration: 7.85398163397
 * /WD002N/joint_limits/upper_joint2/max_position: 2.44346095279
 * /WD002N/joint_limits/upper_joint2/max_velocity: 5.4977
 * /WD002N/joint_limits/upper_joint2/min_position: -2.44346095279
 * /WD002N/joint_limits/upper_joint3/has_acceleration_limits: True
 * /WD002N/joint_limits/upper_joint3/has_position_limits: True
 * /WD002N/joint_limits/upper_joint3/has_velocity_limits: True
 * /WD002N/joint_limits/upper_joint3/max_acceleration: 3.0
 * /WD002N/joint_limits/upper_joint3/max_position: 0.15
 * /WD002N/joint_limits/upper_joint3/max_velocity: 0.255
 * /WD002N/joint_limits/upper_joint3/min_position: 0
[... 101 lines cut ...]
[ERROR] [1679504148.317577855]: [KhiRobotKrnxDriver] [krnx_PrimeRtcCompData] ano:2 [jt]pos:vel:status [1]2.3561:0.0000:0 [2]-2.0942:0.0000:0 [3]0.1200:0.0000:0 [4]-0.2549:0.0000:0 
[ERROR] [1679504148.470216994]: [KhiRobotKrnxDriver] RTC SWITCH turned OFF 0: ano:1
[ INFO] [1679504148.470241777]: [KhiRobotKrnxDriver] State 0: ERROR -> DEACTIVATING
[ INFO] [1679504151.129649072]: [KhiRobotKrnxDriver] State 0: DEACTIVATING -> INACTIVE
[INFO] [1679504151.134627]: Loading controller: duaro_upper_arm_controller
[INFO] [1679504152.938870]: Controller Spawner: Loaded controllers: joint_state_controller, duaro_lower_arm_controller, duaro_upper_arm_controller
[INFO] [1679504153.018751]: Started controllers: joint_state_controller, duaro_lower_arm_controller, duaro_upper_arm_controller
[... cut ...]
[ INFO] [1679504223.779459046]: [KhiRobotKrnxDriver] State 0: DISCONNECTING -> DISCONNECTED
[WARN] [1679504223.923403]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
shutting down processing monitor...
... shutting down processing monitor complete
done
 

Also, it would be great if you could provide the steps to solve this error, because the link you have referenced is just giving the information about what the error is and not how to solve it

siddharthbhurat4, 2023-04-03:
I was able to solve this issue by reducing the velocity limits of each joint. 

Thank you.