Kawasaki-Robotics/khi_robot #83: moveit indicating successful move but real robot not moving

Landon-Getting, 2023-05-17:
I am developing a support package for the CX165L robot. I am able to visualize the robot in rviz, receive correct joint states for the real robot via the /joint_states topic, and plan/execute trajectories in MoveIt. 

However, the real robot does not move despite MoveIt indicating the trajectory was successful. I have attached terminal outputs for the following:

- [rostopic echo cx165l_arm_controller/follow_joint_trajectory/goal](https://github.com/Kawasaki-Robotics/khi_robot/files/11501771/echo.controller.goal.txt)

- [joint states before and after execution](https://github.com/Kawasaki-Robotics/khi_robot/files/11501772/joint.states.txt)

- [output from moveit_planning_execution.launch with a planned and executed trajectory](https://github.com/Kawasaki-Robotics/khi_robot/files/11501773/moveit_planning_execution.output.txt)

This is a current state of the rqt_graph:
[image]

I am happy to provide additional files. @matsui-hiro please let me know if you have any ideas. If successful, I can create a PR to add CX165L functionality if you are interested.

matsui-hiro, 2023-05-19:
I checked the terminal log, but no cause was found.
I would like more information.

If it's OK ,could you upload the launch file of CX165L and tell me how to run it ?
I think source code is [here](https://github.com/Landon-Getting/khi_robot),but can't find a moveit execution launch file.

Also, it is possible that the joint values ware not sent to the controller due to the network environment.
I want your system information.(on native Ubuntu, or WSL, etc.)

matsui-hiro, 2023-05-30:
Thank you for sharing the source code.

I ran the launch file in our environment and the same problem occurred.
After investigation, I found that the cause may be in the [`libkrnx`](https://github.com/Kawasaki-Robotics/khi_robot/tree/master/khi_robot_control/lib) (communication library with the controller).
Depending on the AS system version of the controller, sending joint values may fail. Therefore, please let me know the version of your AS system software.

And I plan to update the library to solve the problem.

Landon-Getting, 2023-05-30:
Hi @matsui-hiro! Thank you for investigating the problem.

The AS system version is **ASEB01010XX3S**. Please let us know if there is anything we can do to help solve the issue.

This is the system environment on our E02 controller:

```
.*=== AS GROUP ===         : **ASEB01010XX3S** 2022/08/02 07:27
.*USER IF AS               : UASE01010XX3S 2022/08/02 07:27
.*USER IF TP               : UTPE01010XX3S 2022/08/02 07:25
.*ARM CONTROL AS           : AASE01010XX3S 2022/08/02 07:26
.*USER IF AS MESSAGE FILE  : MASE010XX3SEN 2022/08/02 07:27
.*USER IF TP MESSAGE FILE  : MTPE010XX3SEN 2022/08/02 07:25
.*ARM DATA FILE            : ARME01010XX3S 2022/08/02 07:25
.*KERNEL                   : _KNL103300000 2016/01/13
.*DRIVER                   : _DRV104800000 2018/02/08
.*RFS                      : _RFS20030X202 2021/08/31
.*=== SERVO GROUP ===      : SVE_08000006C 2021/06/09 14:52
.*ARM CONTROL SERVO        : ASVE08000006C 2021/06/09 14:47
.*SRV DATA FILE            : ASPE08000006C 2021/06/09 14:47
.*ARM CONTROL SERVO FPGA   : ASFE08000000A 2015/04/01 09:29
.*
.*Cpu board type           : 1VA
.*CF                       : 4GB CompactFlash Card 20180831 0000047E
```

matsui-hiro, 2023-06-01:
> The AS system version is ASEB01010XX3S. Please let us know if there is anything we can do to help solve the issue.

Thank you for your cooperation.

`ASEB01010XX3` is  an E Series(OpenAS) software and some versions of E Series(OpenAS) cannot receive joint values with the current library. I will let you know when I have fixed and uploaded the library.

matsui-hiro, 2023-06-02:
>When can we expect the fix to be released? Is there anything we can do to help?

Updates are scheduled by June 9. 
Please confirm the correction in your environment after the update.

matsui-hiro, 2023-06-09:
Hi, @Landon-Getting.
I updated the library. Could you please try again with this? 
When you try with [CX165L](https://github.com/fam-lab/khi_ws/tree/master/src), replace only [khi_robot_control](https://github.com/Kawasaki-Robotics/khi_robot/tree/master/khi_robot_control) pkg.

Landon-Getting, 2023-06-09:
Hi @matsui-hiro!

Thank you for the update. I have replaced the khi_robot_control package in my local workspace. I also recently upgraded to a different machine with Ubuntu 20.04 and ROS Noetic.

Although your change may have worked, I am now encountering different errors during bringup. Please see the [terminal_log.txt](https://github.com/Kawasaki-Robotics/khi_robot/files/11710719/terminal_log.txt).

I've attempted to solve the particular error below by switching via the `/controller_manager/switch_controller service` to `position_controllers/JointGroupPositionController` from the cx165l_controllers.yaml file in the bringup config.
```
[ERROR] [1686353857.803995864]: Could not load controller 'cx165l_arm_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1686353857.804049420]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
```
I will continue troubleshooting the errors and let you know if they are resolved. If you have any solutions, please let me know.

Landon-Getting, 2023-06-12:
Hi @matsui-hiro,

My local machine was missing the joint-trajectory-controller library.
`sudo apt install ros-noetic-joint-trajectory-controller`

After installing, I was able to run the `moveit_planning_execution.launch` file and move the real robot!

Thank you for your help! If you would like, I can help create a pull request for the cx specific directories.

matsui-hiro, 2023-06-16:
Hi, @Landon-Getting.

>After installing, I was able to run the moveit_planning_execution.launch file and move the real robot!

I am glad to hear that you are now able to move cx !

>Thank you for your help! If you would like, I can help create a pull request for the cx specific directories.

For the merge,  need to adjust the acceleration values in [joint_limits.yaml](https://github.com/fam-lab/khi_ws/blob/master/src/khi_cx_description/config/cx165l_joint_limits.yaml) and align the file format with the other robots.  After that work is done I will create a pull request. (the timeing is in plannning)
Thank you for your cooperation !

madelinegannon, 2024-06-11:
Thanks for sharing back @Landon-Getting ... This solved my issue, too:
> My local machine was missing the joint-trajectory-controller library.
> `sudo apt install ros-noetic-joint-trajectory-controller`
>
> After installing, I was able to run the `moveit_planning_execution.launch` file and move the real robot!