Kawasaki-Robotics/khi_robot #90: can't connect real robot(RS007l)

elripunt, 2024-04-11:
hello.
I try to connect rs007l and F60 controller to ros
So, referring to the ConnectingRealRobot.md file, I connected the Ethernet cable to the pc and controller.
And I ran `roslaunch khi_robot_bringup rs013n_bringup.launch ip:=192.168.5.10`

But I got this error message
```
auxapi_login error ret=-8200
asapi_connect error ret=-4099
[ERROR] [1712824106.423495507]: [KhiRobotKrnxDriver] krnx_Open returned -0x1003
[ INFO] [1712824106.423616006]: [KhiRobotKrnxDriver] State 0: CONNECTING -> INIT
[ERROR] [1712824106.423767172]: Failed to open KHI robot
[ INFO] [1712824106.423895855]: [KhiRobotKrnxDriver] State 0: INIT -> DISCONNECTING
[ERROR] [1712824106.424007162]: [KhiRobotKrnxDriver] krnx_Close returned -0x2000
[ INFO] [1712824106.424089355]: [KhiRobotKrnxDriver] destructor
[ERROR] [1712824106.424166824]: [KhiRobotKrnxDriver] krnx_Close returned -0x2000
[ INFO] [1712824106.424244112]: [KhiRobotKrnxDriver] destructor
[khi_robot_control-2] process has died [pid 9677, exit code -11
```
Can you please help me solve this problem?

[ROBOT and PC version]
os: ubuntu20.04
ros: noetic

robot name: rs007l-A001
controller: F60
AS version: ASF_01000001A (Is this version higher than ASF_01000000W??)

thank you.

matsui-hiro, 2024-04-11:
Hi @elripunt,

>And I ran roslaunch khi_robot_bringup rs013n_bringup.launch ip:=192.168.5.10

When you want to connect to rs007l, plsease run **rs007l**_bringup.launch. (Not  rs013n_bringup.launch)
If the error still occurs, could you check for connectivity by ping?

elripunt, 2024-04-11:
sorry i try `roslaunch khi_robot_bringup rs007l_bringup.launch ip:=192.168.5.10`  not rs013n
and i try again  rs007l_bringup.launch but it's the same error message

so I checked with this command `ping 192.168.5.10`

`PING 192.168.5.10 (192.168.5.10) 56(84) bytes of data.
64 bytes from 192.168.5.10: icmp_seq=1 ttl=64 time=0.065 ms
64 bytes from 192.168.5.10: icmp_seq=2 ttl=64 time=0.060 ms
64 bytes from 192.168.5.10: icmp_seq=3 ttl=64 time=0.074 ms
64 bytes from 192.168.5.10: icmp_seq=4 ttl=64 time=0.063 ms
64 bytes from 192.168.5.10: icmp_seq=5 ttl=64 time=0.064 ms
64 bytes from 192.168.5.10: icmp_seq=6 ttl=64 time=0.062 ms
64 bytes from 192.168.5.10: icmp_seq=7 ttl=64 time=0.080 ms
64 bytes from 192.168.5.10: icmp_seq=8 ttl=64 time=0.063 ms
64 bytes from 192.168.5.10: icmp_seq=9 ttl=64 time=0.064 ms
64 bytes from 192.168.5.10: icmp_seq=10 ttl=64 time=0.055 ms
64 bytes from 192.168.5.10: icmp_seq=11 ttl=64 time=0.080 ms
64 bytes from 192.168.5.10: icmp_seq=12 ttl=64 time=0.066 ms
64 bytes from 192.168.5.10: icmp_seq=13 ttl=64 time=0.064 ms
64 bytes from 192.168.5.10: icmp_seq=14 ttl=64 time=0.069 ms`

closed 2024-04-17 by elripunt

elripunt, 2024-04-17:
pc and the controller's ip should not be set the same
 It's my mistake
Thank you!