Kawasaki-Robotics/khi_robot #91: ROS 2 support?

jdham-iastate, 2024-08-30:
Hi, are there plans for ROS 2 integration? I know the question has been asked prior, but I want to bring the topic to relevancy again.

MasaoAramoto, 2024-09-03:
We are currently developing the ROS2 Humber version. However, the release schedule has not been decided yet.

jdham-iastate, 2024-10-13:
Any update on the ROS2 driver? We're eager to get an implementation in ROS2.

jdham-iastate, 2024-11-20:
Hey @MasaoAramoto we took a stab at integrating KRNX with ros2_control in Humble. You're welcome to see our driver at [ham-lab-isu/khi2cpp_hw](https://github.com/ham-lab-isu/khi2cpp_hw) and [ham-lab-isu/khi2cpp_hw_description](https://github.com/ham-lab-isu/khi2cpp_hw_description). **We have it running on our CX165L and is working well. We have motion and have it integrated with [Scan 'N Plan](https://rosindustrial.org/scan-n-plan) from SwRI**

dantetemplar, 2025-12-10:
Should be closed.