Kawasaki-Robotics/khi_robot #92: Need ROS2 Humble driver.

wglockner, 2024-08-30:
Currently, most packages being developed are in ROS2 and there does not seem to be any support for ROS2 for this driver.  Our lab has moved all development to humble as it is the latest LTS version.  Noetic can not be run on Ubuntu 22.04 so utilizing this driver would not be possible without a docker image, which also limits future development.

MasaoAramoto, 2024-09-03:
We are currently developing the ROS2 Humber version. However, the release schedule has not been decided yet.

kurita-taisuke, 2025-08-08:
Apologies for the delay. The ROS2 driver has been released and is compatible with Humble.  
https://github.com/Kawasaki-Robotics/khi_ros2

dantetemplar, 2025-12-10:
Should be closed.