Kawasaki-Robotics/khi_robot #96: ROS2 migration

pawelir, 2025-04-07:
Hi Kawasaki Team,

I've seen your responses to existing issues and understand you're actively working on a ROS 2 Kawasaki drive!

I was wondering if you could share an update on the migration status. Do you have a rough estimate for when the ROS 2 version might be released?

Also, would you consider making your development branch public, even if it's still a work in progress? I believe the community would appreciate early access - it might help accelerate development or enable some of your users in the meantime.

wglockner, 2025-07-02:
Any updates on this?

pawelir, 2025-07-03:
Nope, this repo is dead.

CC @matsui-hiro @ohno-atsushi

MasaoAramoto, 2025-07-03:
Sorry for late reply.
We have aleady prepared the code for ROS2 hamble.
Now we are trying to adapt to ROS2 coding style and so on.
I think we can release it here in a few weeks.

pawelir, 2025-07-04:
Hi @MasaoAramoto, thanks for your answer! Would you consider making the prerelease version public? I'm pretty sure the community will be able to help with applying ROS 2 best practices to it.

pawelir, 2025-08-13:
Thank you @kurita-taisuke, that's great news!